hexadecimalDinosaur/octoprint-cli #13: Command Usage Wrong

CommanderRedYT, 2021-01-03:
**Describe the bug**
In the README.md it says `octoprint-cli ...` , but you would use `python3 octoprint-cli.py ..`.

**Additional context**
Would fix it with an install script (maybe .sh script) that 1. checks for dependencies and 2. maybe edits .bashrc for alias? (Not that good at programming so I wouldn't take this as the solution..!

UserBlackBox, 2021-01-03:
Are you using Windows or Linux? The reason I put it that way in the README.md is because on Linux if you symlink it or place it at the path `~/.local/bin/octoprint-cli` you can run it in the terminal without the python command by using the shebang.

Having an install script is a good idea. I can probably make one to download dependencies and create symlinks in `~/.local/bin`

CommanderRedYT, 2021-01-04:
Using Ubuntu 20.04. For the install script you could do it like NODE-RED:
[image]

(Its just an file containing shell script stored in the repo)

UserBlackBox, 2021-01-05:
I decided that I won't be making an install script and instead will be uploading the script to PyPI so that it can be installed and updated with `pip`. This should help with downloading dependencies and setting up the program to run from the terminal without needing to run python. I will update the README.md with instructions on installing using `pip` soon.

UserBlackBox, 2021-01-05:
octoprint-cli has been packaged and upload to [PyPI](https://pypi.org/project/octoprint-cli/) which helps install the program and dependencies using `pip`

```bash
pip install octoprint-cli --user
```

The program can also be updated in the future using `pip install --upgrade octoprint-cli --user`

closed 2021-01-05 by UserBlackBox