hexadecimalDinosaur/octoprint-cli #14: Error with argument "contiuous"

CommanderRedYT, 2021-01-03:
**Describe the bug**
Script crashes when using `octoprint-cli continuous`

**To Reproduce**
Steps to reproduce the behavior:
1. Turn off heatbed in octoprint settings
2. For Nr1 goto Settings -> Printer profiles -> Edit your profile -> Printbed & Volume -> Make sure heatbed is turned off

**Expected behavior**
Script detects that heatbed is turned off

**Screenshots**
[image]

**Additional context**
Reproduce-steps could be wrong name, have my instance in german language so I just assume it is what it would translate to

UserBlackBox, 2021-01-03:
I am looking into this issue but I can't find the option in OctoPrint to turn off the print bed. What version of OctoPrint are you using and what printer do you have?

Just for reference this is how it looks like on OctoPrint 1.5.2 on OctoPi 0.17.0
[image]

CommanderRedYT, 2021-01-04:
Its right below Origin! (Heated Bed-Checkmark)

[image]

UserBlackBox, 2021-01-04:
Ok, I may have misread your issue and thought there was an option for turning off the print bed. What printer are you using? I think the issue may come from using a printer that doesn't continue to report bed temperatures when the bed is turned off. If you tell me what printer you are using I would have a better idea of why this issue is occuring.

CommanderRedYT, 2021-01-04:
`ok T:21.2 /0.0 B:0.0 /0.0 T0:21.2 /0.0 @:0 B@:0` - that would be the response of an **M105**-Command.

My printer has an option for a heated bed, but i didn't buy it. But I don't know exactly if the issue is the M105-response or that I deactivated heated bed in the octoprint settings.
What I thought was that the API would send an empty string / the string would end where the heated bed-message would be. So the program crashes because of an empty string... 

So idk if you know what I mean but when I say octoprint settings I am referring to the checkmark I marked red in my comment above

UserBlackBox, 2021-01-04:
So the issue at the moment is that octoprint-cli is made for printers with a heated bed but on a printer without one the API request returns a 409 Conflict which breaks the program. I'm pretty sure this not just an issue in continuous but can be found in the `temp status` commands and `print status` command. I'll try to work on it sometime this week to resolve this issue.

UserBlackBox, 2021-01-05:
This issue should be fixed as of release [v3.2.0](https://github.com/UserBlackBox/octoprint-cli/releases/tag/v3.2.0) so I will be closing this issue. If you still run into issues with this please reopen this issue and let me know.

closed 2021-01-05 by UserBlackBox