hexadecimalDinosaur/octoprint-cli #16: TypeError when run in continuous mode or print status mode

geoffdavis, 2021-01-10:
**Describe the bug**
Running `octoprint-cli continuous` or `octoprint-cli print status` results in a traceback with a TypeError

**To Reproduce**
Steps to reproduce the behavior:
1. run `octoprint-cli`

**Expected behavior**
It should display valid data and keep running.

**Screenshots**
```
pi@mpmd-pi:~ $ octoprint-cli continuous
Loaded File: Shape-Box_N150um_PLA_MPMD.gcode
Progress: 52.39%
Traceback (most recent call last):
  File "/home/pi/.local/bin/octoprint-cli", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 759, in main
    options.func(options)
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 188, in continuous
    jobPrint()
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 164, in jobPrint
    print(colored("Estimated Print Time: ", attrs=['bold']) + str(datetime.timedelta(seconds=data['job']['estimatedPrintTime'])).split(".")[0])
TypeError: unsupported type for timedelta seconds component: NoneType
```

```
pi@mpmd-pi:~ $ octoprint-cli print status
Printing
File: Shape-Box_N150um_PLA_MPMD.gcode
Traceback (most recent call last):
  File "/home/pi/.local/bin/octoprint-cli", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 759, in main
    options.func(options)
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 299, in print_status
    jobPrint()
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 285, in jobPrint
    print(colored("Estimated Total Print Time: ", attrs=['bold']) + caller.getTotalTime())
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 86, in getTotalTime
    hours = int(time//3600)
TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'
```

**Environment Information:**
 - OS: Octopi 0.17 based on Raspbian GNU/Linux 10 (buster) with Octoprint running python3 via the Blog entry conversion script.
 - Version: 3.2.0.post3
 - Python Version: Python 3.7.3
 - Dependency versions: requests-2.25.1, termcolor-1.1.0
 - OctoPrint version: 1.5.2, running under python3
 - 3D Printer Model: Monoprice MiniDelta with MPMD_Marlin-1.1.x firmware

**Additional context**
Other commands seem to work ok. Just not `octoprint-cli print status` or `octoprint-cli continuous`

```
pi@mpmd-pi:~ $ octoprint-cli temp status
Extruder Temp: 194.97°C
Extruder Target: 195.0°C
Bed Temp: 60.13°C
Bed Target: 60.0°C
pi@mpmd-pi:~ $ octoprint-cli connection status
Printing
Printer Profile: MPMD
Port: /dev/ttyACM0
Baudrate: 115200
pi@mpmd-pi:~ $ octoprint-cli system status
usage: octoprint-cli system [-h] {restart,restart-safe,reboot,shutdown} ...
octoprint-cli system: error: invalid choice: 'status' (choose from 'restart', 'restart-safe', 'reboot', 'shutdown')
pi@mpmd-pi:~ $ octoprint-cli files status
usage: octoprint-cli files [-h] {list,info,upload} ...
octoprint-cli files: error: invalid choice: 'status' (choose from 'list', 'info', 'upload')
pi@mpmd-pi:~ $ octoprint-cli files list
NAME                                 TYPE           SIZE
Landy                                folder
calibration                          folder
2xtire_mpmd_N150um_PLA_MPMD.gcode    machinecode    8.43 MB
Shape-Box_N150um_PLA_MPMD.gcode      machinecode    0.12 MB
tire_mpmd_n150um_pla_mpmd.gcode      machinecode    4.29 MB
wheel-rim_n150um_pla_mpmd.gcode      machinecode    4.45 MB

Free space: 25.664 GB
```

UserBlackBox, 2021-01-10:
I don't know why the `/api/job` call is giving a None instead of an integer in the API call. Can you send me the output of this command?
```bash
curl -H 'X-API-Key: YourAPIKeyHere' http://addressHere/api/job
```

UserBlackBox, 2021-01-21:
Oh, I see the issue. It is showing up for me as well. I think it is a change to the API of OctoPrint.

UserBlackBox, 2021-01-21:
Should be fixed in [3.2.1](https://pypi.org/project/octoprint-cli/3.2.1/) @geoffdavis

closed 2021-01-21 by UserBlackBox