hexadecimalDinosaur/octoprint-cli #19: TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'

e77, 2021-01-30:
When octoprint process a very large gcode file to be ready for print (this one is 52.8mb) there appears to be an overflow error.

Continuous mode works fine until you either attempt to star the print, or you load the gcode viewer.
It's at this point that the layer count is generated, along with the expected print time.  582 layers, approx 7 hours.

Doesn't happen with smaller files - still testing, but so far anything over 40mb causes the error output below:

Traceback (most recent call last):
  File "/home/pi/.local/bin/octoprint-cli", line 11, in <module>
    load_entry_point('octoprint-cli==3.2.2', 'console_scripts', 'octoprint-cli')()
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.2.2-py3.7.egg/octoprint_cli/__main__.py", line 669, in main
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.2.2-py3.7.egg/octoprint_cli/__main__.py", line 91, in continuous
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.2.2-py3.7.egg/octoprint_cli/api.py", line 68, in getTotalTime
TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'

*

UserBlackBox, 2021-01-30:
Thanks for the notice, I haven't run into this issue in the past (though that's probably because I only do relatively small prints). It appears that the OctoPrint API does not supply a value for `estimatedPrintTime` while files are being processed based off the information you have provided me. I will try to recreate this issue myself and look at the API data. What version of OctoPrint are you using?

PitBarber, 2021-04-19:
I had same error yesterday, seems to be when you press "start at upload" in your favourite slic3r... if you upload the file and press print later at octoprint server error doesn't appear.

UserBlackBox, 2021-04-28:
Yeah, I have run into this error myself once. It seems that the server sends incomplete API data when its processing files. I need to try to trigger this issue for testing and make a catch for this incomplete API data.