hexadecimalDinosaur/octoprint-cli #21: "Octoprint server cannot be reached

willpuckett, 2021-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment Information:**
 - OS: [e.g. iOS]
 - Version [e.g. 22]
 - Python Version: [e.g. 3.8]
 - Dependency versions
 - OctoPrint version: 
 - 3D Printer Model: 

**Additional context**
Add any other context about the problem here.

willpuckett, 2021-03-10:
Thanks for responding so fast!

curl output:
{"api":"0.1","server":"1.5.3","text":"OctoPrint 1.5.3"}

Really looking forward to getting this figured out looks like you've done some awesome work :)

willpuckett, 2021-03-10:
Just poked around a little further...

When I supply ServerAddress and ApiKey with no quotes, I get: "Invalid or Missing Arguments" followed by the help message. When I use soft or hard quotes, I get "Octoprint server cannot be reached"....

UserBlackBox, 2021-03-10:
> Just poked around a little further...
> 
> When I supply ServerAddress and ApiKey with no quotes, I get: "Invalid or Missing Arguments" followed by the help message. When I use soft or hard quotes, I get "Octoprint server cannot be reached"....

The configuration file shouldn't have quotes. Should look something like the image below. `Invalid or Missing Arguments` should only be showing up if the program arguments used are incorrect. Can you remove the quotes and try `octoprint-cli version` and tell me if the issue persists?

[image]

willpuckett, 2021-03-10:
Ahh, user error it is. I was expecting an interactive session when I executed octoprint-cli, but I now see I need to issue a sub command at runtime.

Thanks so much.

closed 2021-03-10 by willpuckett