hexadecimalDinosaur/octoprint-cli #23: error while printing

jzamero, 2021-04-14:
keep getting this errors while printing

pi@octopi:~/octoprint-cli $ octoprint-cli  continuous
Printing
Loaded File: ASX1_Part6.gcode
Progress: 15.77%
Traceback (most recent call last):
  File "/home/pi/.local/bin/octoprint-cli", line 11, in <module>
    load_entry_point('octoprint-cli==3.3.0', 'console_scripts', 'octoprint-cli')()
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.0-py3.7.egg/octoprint_cli/__main__.py", line 890, in main
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.0-py3.7.egg/octoprint_cli/__main__.py", line 138, in continuous
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.0-py3.7.egg/octoprint_cli/__main__.py", line 92, in jobPrint
TypeError: unsupported type for timedelta seconds component: NoneType

UserBlackBox, 2021-04-28:
What OS, Python version, and OctoPrint server version are you using?

If you have curl can you try sending me the output of this command during one of your prints?
```bash
curl -H 'X-API-Key: API_KEY_HERE' http://SERVER_ADDRESS_HERE/api/job
```

Also does this issue show up every time you print?

jzamero, 2021-04-30:
Hi
thank you for replying
following is the output of the command, it is happened all the time

{"job":{"averagePrintTime":null,"estimatedPrintTime":null,"filament":null,"file":{"date":1619765984,"display":"Hakko
Station
Cover_0.2mm_PLA_X1_1h33m.gcode","name":"Hakko_Station_Cover_0.2mm_PLA_X1_1h33m.gcode","origin":"local","path":"Hakko_Station_Cover_0.2mm_PLA_X1_1h33m.gcode","size":1702042},"lastPrintTime":null,"user":"jacky"},"progress":{"completion":2.9258384928221512,"filepos":49799,"printTime":0,"printTimeLeft":null,"printTimeLeftOrigin":"linear"},"state":"Pausing"}

Jacky

UserBlackBox, 2021-04-30:
This issue and #19 are both caused by a lack of complete API data returned from the server. I am unsure what causes this server side for this specific issue so I will be making a catch for incomplete data. A perfect solution to this problem would have to be implemented server-side but this solution should let you at least view what data was received from the server.

UserBlackBox, 2021-05-03:
Can you test release [v.3.3.1](https://github.com/UserBlackBox/octoprint-cli/releases/tag/v3.3.1). I wrote a catch for the missing data.

jzamero, 2021-05-03:
I will

jzamero, 2021-05-07:
i've tried and now it completely crash

UserBlackBox, 2021-05-07:
Can you send the output of the program or error messages? Can you also make sure you have updated to 3.3.1 with `pip install octoprint-cli==3.3.1`?

jzamero, 2021-05-07:
Loaded File: Left_-_lanyard_attachment_0.16mm_PLA_X1_9h55m.gcode
Progress: 54.33%
Traceback (most recent call last):
  File "/home/pi/.local/bin/octoprint-cli", line 11, in <module>
    load_entry_point('octoprint-cli==3.3.0', 'console_scripts',
'octoprint-cli')()
  File
"/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.0-py3.7.egg/octoprint_cli/__main__.py",
line 890, in main
  File
"/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.0-py3.7.egg/octoprint_cli/__main__.py",
line 138, in continuous
  File
"/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.0-py3.7.egg/octoprint_cli/__main__.py",
line 92, in jobPrint
TypeError: unsupported type for timedelta seconds component: NoneType

UserBlackBox, 2021-05-07:
It appears from that error message that its still using 3.3.0. Can you update the program to 3.3.1 using this command?
```bash
pip install octoprint-cli --upgrade
```

jzamero, 2021-05-07:
now it's working
thank you

closed 2021-05-07 by UserBlackBox