hexadecimalDinosaur/octoprint-cli #24: TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'

jacksnodgrass, 2021-06-08:
**Describe the bug**
TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'

This fails on BOTH fedora 34 and my raspberry pi that is running octopi

**To Reproduce**
octoprint-cli print status
```

octoprint-cli print status
Printing
File: PIKVM_Case_Bottom_3h55m_0.16mm_205C_PLA_ENDER3.gcode
Traceback (most recent call last):
  File "/opt/anaconda3/bin/octoprint-cli", line 8, in <module>
    sys.exit(main())
  File "/opt/anaconda3/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 896, in main
    options.func(options)
  File "/opt/anaconda3/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 290, in print_status
    jobPrint()
  File "/opt/anaconda3/lib/python3.7/site-packages/octoprint_cli/__main__.py", line 272, in jobPrint
    attrs=['bold']) + caller.getTotalTime())
  File "/opt/anaconda3/lib/python3.7/site-packages/octoprint_cli/api.py", line 90, in getTotalTime
    hours = int(time//3600)
TypeError: unsupported operand type(s) for //: 'NoneType' and 'int'
```

**Expected behavior**
not fail... give good results. 

**Environment Information:**
 - OS: Fedora 34 and Raspberry Pi running octopi 
 - Version 3.3.1
 - Python Version: 3.7.6 ( fedora 34) 
 - Python Version: 3.7.3 ( raspberrry pi ) 
 - Dependency versions
 - OctoPrint version: 1.6.1
 - 3D Printer Model: Ender 3

**Additional context**
other commands work fine: 
octoprint-cli files list
octoprint-cli continuous
octoprint-cli version
... the octoprint-cli print status is the only command that I know fails... but I didn't run all the commands.

UserBlackBox, 2021-06-08:
This seems to be a similar issue to #23 but on a different subcommand. Can you try sending the output of `curl -H 'X-API-Key: API_KEY_HERE' http://SERVER_ADDRESS_HERE/api/job` if this happens everytime you print?

jacksnodgrass, 2021-06-08:
I get a lot of nulls... 
curl -H 'X-API-Key: my-super-secret-api-key' http://127.0.0.1/api/job
{"job":{"averagePrintTime":null,"estimatedPrintTime":null,"filament":null,"file":{"date":1623156764,"display":"PIKVM_Case_Bottom_3h55m_0.16mm_205C_PLA_ENDER3.gcode","name":"PIKVM_Case_Bottom_3h55m_0.16mm_205C_PLA_ENDER3.gcode","origin":"local","path":"PIKVM_Case_Bottom_3h55m_0.16mm_205C_PLA_ENDER3.gcode","size":3149876},"lastPrintTime":null,"user":"admin"},"progress":{"completion":99.20904822920014,"filepos":3124962,"printTime":13554,"printTimeLeft":35,"printTimeLeftOrigin":"estimate"},"state":"Printing"}

Currently according to the 'continuous' status, I am 99.64% complete.... 

- jack

UserBlackBox, 2021-06-08:
Can you try replacing `/opt/anaconda3/lib/python3.7/site-packages/octoprint_cli/api.py` with the updated copy in 432ccfc. If this fixes your issue then I'll create a new release with the fix.

jacksnodgrass, 2021-06-08:
that new api.py worked... but I can't see really what you changed in there. It looks like you changed: 
                  (self.address+target, data, request.status_code),file=stderr)
                  to
                 (self.address+target, data, request.status_code),
                  file=stderr)

... how did that fix this? educate me....

UserBlackBox, 2021-06-08:
The change you saw was a formatting change that I did to fix a line that was too long. That change was completely unrelated with your issue. The fix is on line [76](https://github.com/UserBlackBox/octoprint-cli/blob/77ae509a6c872cc2893dbb5ceaf177d381eb94ee/octoprint_cli/api.py#L76) and [93](https://github.com/UserBlackBox/octoprint-cli/blob/77ae509a6c872cc2893dbb5ceaf177d381eb94ee/octoprint_cli/api.py#L93) where I added this to `getTimeLeft()` and `getTotalTime()`.
```python
if not time:
    return "unavailable"
```

closed 2021-06-08 by UserBlackBox