hexadecimalDinosaur/octoprint-cli #25: TypeError: Expected maxsize to be an integer or None

testbughub, 2021-07-11:
**Describe the bug**
I get this TypeError when I try using it;
TypeError: Expected maxsize to be an integer or None
Not sure if I missed something, but it worked fine on my previous Pi (I have upgraded to a Pi 4 from 3).

<details>
  <summary>Full output</summary>
  
```
pi@octopi:~/.config $ octoprint-cli version
Traceback (most recent call last):
  File "/home/pi/.local/bin/octoprint-cli", line 33, in <module>
    sys.exit(load_entry_point('octoprint-cli==3.3.2', 'console_scripts', 'octoprint-cli')())
  File "/home/pi/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 474, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/pi/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2846, in load_entry_point
    return ep.load()
  File "/home/pi/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2450, in load
    return self.resolve()
  File "/home/pi/.local/lib/python3.7/site-packages/pkg_resources/__init__.py", line 2456, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.2-py3.7.egg/octoprint_cli/__main__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 668, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 638, in _load_backward_compatible
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.2-py3.7.egg/octoprint_cli/api.py", line 6, in <module>
  File "/home/pi/.local/lib/python3.7/site-packages/octoprint_cli-3.3.2-py3.7.egg/octoprint_cli/api.py", line 19, in api
  File "/usr/lib/python3.7/functools.py", line 477, in lru_cache
    raise TypeError('Expected maxsize to be an integer or None')
TypeError: Expected maxsize to be an integer or None
```
  
</details>

**To Reproduce**
Steps to reproduce the behavior:
1. git clone
2. `python3 setup.py install --user`
3. `octoprint-cli version`

**Expected behavior**
No errors

**Environment Information:**
 - OS: Latest OctoPi image - Version 0.18.0, running on a Pi 4 Model B Rev 1.1
 - Version: git
 - Python Version: 3.7
 - Dependency versions: termcolor 1.1.0 + requests 2.25.1
 - OctoPrint version: 1.6.1
 - 3D Printer Model: Creality Ender 3 Pro

UserBlackBox, 2021-07-11:
This is an issue with the lru caching that I was adding. This version isn't released so you should install the stable release from pip or checkout `v3.2.2` in git

testbughub, 2021-07-12:
Gotcha, thanks for the info!

closed 2021-07-12 by testbughub